t-oster/VisiCut #663: Download page outdated

mgmax, 2023-01-15:
https://download.visicut.org/ doesn't have the latest commit even after two hours of waiting time.

mgmax, 2023-01-15:
@t-oster it would also be good if you could check whether the build container is up to date. It looks like in the latest build the logo wasn't rendered correctly, though this should have been fixed in the latest VisicutBuilder.

t-oster, 2023-01-15:
I triggered the build. There is no new docker image for the builder but I have to check tomorrow how to improve this

mgmax, 2023-04-25:
In the latest build, the docker image is still outdated, which makes the logo look suboptimal.

t-oster, 2023-04-25:
The gitlab mirror (https://gitlab.com/t-oster/visicutbuildservice) was outdated. I am building the image right now, should be fixed soon.

t-oster, 2023-04-25:
does not seem to build https://gitlab.com/t-oster/visicutbuildservice/-/jobs/4177019752

t-oster, 2023-04-25:
nevermind, seems to be a space issue on gitlab (the image is 1.4GB). I built it locally and pushed it.
I did a fresh build of the 1.9-173 versions.

closed 2023-04-25 by t-oster