t-oster/VisiCut #664: inkscape viscut issues

hirilpatel, 2023-01-23:
Hello,

I dont know if this issue has already been reported but, after a while visicut extension from inkscape stopped working. I am not a computer person here so can someone help me out?

Here is the photo of error message. Please keep in mind that i didn't updated not the computer not the inkscape or visicut.
[image]

mgmax, 2023-01-23:
Well, if you changed nothing, then it can't stop working :-D

Please try reinstalling the Inkscape extension using the Visicut menu (Extras - Install Inkscape extension).

Does the problem also happen on another computer?

mgmax, 2023-02-18:
Closing because there was no response. Please respond if you have more information.

closed 2023-02-18 by mgmax

hirilpatel, 2023-02-18:
Hello,

Sorry for no response. I am not currently in the office and won't be till
end of march hence can't do what you suggest. Can you please keep this
issue open?

I can only check after march