t-oster/VisiCut #665: command-line access to visicut?

valkyriesavage, 2023-01-27:
I can't find anything in the manuals about this, but is there a way to summon visicut from the commandline? I'd like to be able to give it a file with cut parameters, an svg to cut, a laser bed location to cut it at, and whatever is necessary to talk with my laser; this would help with fiddling with settings.

valkyriesavage, 2023-01-30:
ok, I found the docs you mentioned, but it seems that it's ignoring my arguments on commandline (I'm using the [instructions I found](https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline)).

[image]

I did try to do platform-independent installation (it's in my Applications folder), but my computer is not finding a `visicut` command in terminal.

t-oster, 2023-01-30:
Maybe the VisiCut.MacOS does not pass through the command line arguments. Open it in a text editor and add `"$@"` in the line with VisiCut.jar

valkyriesavage, 2023-01-30:
[image]
it seems to already have that?

mgmax, 2023-01-30:
On Linux `visicut -h` shows the help and then exits. The Linux launcher does something similar.
So I guess the script you are showing is not the one that is actually called.

valkyriesavage, 2023-01-31:
[image]

t-oster, 2023-01-31:
Could you try to execute the java line directly (replacing $DIR with '.' and $@ with -h?

valkyriesavage, 2023-02-01:
[image]

It now just exits, but doesn't seem to print to STDOUT. I also tried removing all the quotes, and the same thing happened:
[image]

valkyriesavage, 2023-02-01:
hmm... ok, it seems that I don't have a java runtime, and running that `java` command bare gets me complaints. do I have to have that to run on commandline? I'll go install it and try that.

valkyriesavage, 2023-02-01:
🤦🏼‍♀️ now I'm getting an error that the jar file shipped in the release is too new for my fresh java installation
[image]
I'll go set up the dev compilation stuff

valkyriesavage, 2023-02-01:
NetBeans is not cooperating at all, so fingers crossed there is a solution that isn't "compile from scratch to run on commandline"

t-oster, 2023-02-01:
Does it work with the open command (second line without java?)

t-oster, 2023-02-01:
The correct java version is included in the osx version

mgmax, 2023-12-24:
Closing due to lack of response. Feel free to answer and reopen if you can provide more information.

closed 2023-12-24 by mgmax