t-oster/VisiCut #667: IO Error when trying to send job to GBRL

amroldan, 2023-02-28:
I have an issue similar to that one IO Error when trying to send the job to Smoothieboard  (https://github.com/t-oster/VisiCut/issues/359)

I have an Arduino One with GBRL 1.1 with the same bootloader as an Arduino Nano.

[image]

Visicut can Execute and connect properly with Arduino Uno (original one).

But when I use the Arduino Nano (with CH340) with the same bootloader and GBRL version as the Arduino Uno, Visicut can not connect and download the information.

Both Arduino Uno and Nano can work perfectly with GBRL Panel. 

I have tried to change Serial speed and the milliseconds to wait for an answer. The communication starts but does not continue and at the end, I receive: IO Error from COM5: Underlying input stream returned zero bytes.

Any idea to solve the communication issue ?

amroldan, 2023-02-28:
I have made other test. 
If I use an Arduino Uno with CH340 that also works with GBRL Panel, it doesn´t work with Visicut.

I think there is an issue with CH340 speed communication or a similar problem.

mgmax, 2023-06-09:
This error looks similar to https://github.com/t-oster/LibLaserCut/issues/170 . Looks like VisiCut is doing something slightly wrong regarding RTS/CTS settings, timing or similar. Please try to get a detailed log of the communication and port configuration for both tools (VisiCut and GRBL-panel), so that we can find the difference and fix it. Some possibilities for debugging on Windows can be found in the above link, and there are certainly other tools for serial port sniffing/tracing on other operating systems.