t-oster/VisiCut #669: Add new IRC channel for communication

TheAssassin, 2023-05-24:
Since Freenode unfortunately was killed completely off due to a hostile takeover, I'd like to suggest setting up a new IRC channel on [Libera.Chat](https://libera.chat/) to create a community again outside of GitHub.

*(I'm not sure whether the Freenode channel is still active (can't imagine, but also won't check), but the website still advertises it.)*

mgmax, 2023-05-28:
I guess the old chat is dead, currently nobody was in the channel.

Many projects have moved away from IRC to some web-based stuff. For example, Inkscape has switched to RocketChat https://chat.inkscape.org/ with some bridging to IRC for oldschool developers ;-) . The bonus of RocketChat is that it supports offline messages and email notifications.

The VisiCut GitHub discussions forum seems to be sparingly used, but is better than nothing. https://github.com/t-oster/VisiCut/discussions  I just added a link to visicut.org to make it more visible.

If nobody disagrees, then I would suggest that you set up whatever chat you want and update the link. (The website is also a Git repo, just use the "Edit" link at the bottom.) Then we can see if the chat stays alive.

TheAssassin, 2023-06-02:
I registered `#visicut` on the Libera.Chat IRC network. I'll transfer it upon request, if you want me to, so please just let me know.