t-oster/VisiCut #670: Set up CI/CD completely with GitHub actions

TheAssassin, 2023-05-24:
To make development easier and allow contributors to test their changes in the real world before they send PRs, I would like to move CI/CD completely to GitHub actions. This would further allow for continuously deploying VisiCut in the form of a moving tag for which a prerelease can be created (this is what many AppImages do for instance, and the AppImage community provides tools that make this very easy).

In case you feel uncomfortable sharing secrets with GitHub (e.g., for code signing), you could set up a custom runner or perform the final steps on other private infrastructure. Both GitHub actions and any other CI/CD system should share the same scripts.

mgmax, 2023-05-25:
I agree. Feel free to do so. 🙂 https://github.com/t-oster/VisicutBuilder should be a good starting point for the build environment/Docker container.

The visicut repo already contains "distribute.sh" which does 95% of all the building. It only lacks a few details such as the bundled Windows EXE for autotrace. These remaining 5% should be moved over from VisicutBuilder.

TheAssassin, 2023-05-25:
I already rewrote most of `distribute.sh` (just didn't get to creating a PR) to make it more usable and include the AppImage build in it.

TheAssassin, 2023-05-28:
#671 introduced a prototype Docker based build system based on the old AppImage-specific Dockerfile. Ideally, this will replace other Docker images to avoid redundancies.

mgmax, 2023-06-28:
related: https://github.com/t-oster/VisiCut/pull/680

TheAssassin, 2024-03-07:
PR available in #710.

PR #710 (2024-03-24): Release with GitHub actions
Fixes #670.

This PR introduces a GitHub actions workflow to create releases on GitHub with all the available binaries.

It is intended to replace the external VisiCut build server and make it easier for contributors to maintain their own builds without the need for external infrastructure and complicated setup procedures.

Please note that the second commit is a workaround for an NSIS bug. The generated license file contains non-ASCII characters (it is UTF-8 encoded). I cannot make NSIS accept the file as-is, it claims it violates codepage 0 (to my understanding, t [... cut]
closed 2024-03-24 by mgmax