t-oster/VisiCut #675: visicut.model.graphicelements.ImportException: java.lang.NullPointerException

nicolas-raoul, 2023-05-31:
I get a `NullPointerException` when importing this SVG:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   width="210mm"
   height="297mm"
   viewBox="0 0 210 297"
   version="1.1"
   id="svg5"
   inkscape:version="1.2.2 (b0a8486, 2022-12-01)"
   sodipodi:docname="nicolas-katakana.svg"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:svg="http://www.w3.org/2000/svg">
  <sodipodi:namedview
     id="namedview7"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:showpageshadow="2"
     inkscape:pageopacity="0.0"
     inkscape:pagecheckerboard="0"
     inkscape:deskcolor="#d1d1d1"
     inkscape:document-units="mm"
     showgrid="false"
     inkscape:zoom="0.60572619"
     inkscape:cx="176.64747"
     inkscape:cy="560.48427"
     inkscape:window-width="1597"
     inkscape:window-height="997"
     inkscape:window-x="0"
     inkscape:window-y="25"
     inkscape:window-maximized="1"
     inkscape:current-layer="layer1" />
  <defs
     id="defs2">
    <rect
       x="295.51306"
       y="774.27723"
       width="237.73118"
       height="234.42935"
       id="rect773" />
  </defs>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1">
    <text
       xml:space="preserve"
       style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:98.7997px;line-height:0;font-family:TakaoMincho;-inkscape-font-specification:'TakaoMincho, Normal';font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-variant-east-asian:normal;letter-spacing:0px;fill:#000000;fill-opacity:1;stroke:none;stroke-width:2.47"
       x="107.3364"
       y="7.4216676"
       id="text591"><tspan
         sodipodi:role="line"
         id="tspan769"
         style="writing-mode:vertical-lr;stroke-width:2.47"
         x="107.3364"
         y="7.4216676"
         dy="0 -20 -5"
         rotate="0 0">ニコラ</tspan></text>
  </g>
</svg>
```
[image]
Did I do something wrong? Example files such as `bendable-material.svg` can be imported and displayed fine. I am using Java 11.
Thanks!

t-oster, 2023-05-31:
Hi,
please start VisiCut from the command line and paste the full stack trace of the exception. https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline

t-oster, 2023-05-31:
but here is the trace:
```
java.lang.ArrayIndexOutOfBoundsException: Index 2 out of bounds for length 2
	at com.kitfox.svg.Tspan.appendToShape(Tspan.java:282)
	at com.kitfox.svg.Text.buildText(Text.java:362)
	at com.kitfox.svg.Text.build(Text.java:266)
	at com.kitfox.svg.Text.updateTime(Text.java:610)
	at com.kitfox.svg.Group.updateTime(Group.java:313)
	at com.kitfox.svg.Group.updateTime(Group.java:313)
	at com.kitfox.svg.SVGRoot.updateTime(SVGRoot.java:409)
	at com.kitfox.svg.SVGDiagram.updateTime(SVGDiagram.java:243)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:610)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:497)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:482)
	at de.thomas_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:137)
	at de.thomas_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:370)
	at de.thomas_oster.visicut.model.graphicelements.AbstractImporter.importFile(AbstractImporter.java:37)
	at de.thomas_oster.visicut.model.graphicelements.GraphicFileImporter.importFile(GraphicFileImporter.java:88)
	at de.thomas_oster.visicut.VisicutModel.loadGraphicFile(VisicutModel.java:706)
	at de.thomas_oster.visicut.VisicutModel.loadFile(VisicutModel.java:491)
	at de.thomas_oster.visicut.gui.MainView.loadFileReal(MainView.java:1869)
	at de.thomas_oster.visicut.gui.MainView$56.run(MainView.java:1769)
```

seems as if there is a problem with parsing of the tspan element

t-oster, 2023-05-31:
If you convert the text to a path with inkscape, it works:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   width="210mm"
   height="297mm"
   viewBox="0 0 210 297"
   version="1.1"
   id="svg5"
   inkscape:version="1.2.2 (b0a8486541, 2022-12-01)"
   sodipodi:docname="test.svg"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:svg="http://www.w3.org/2000/svg">
  <sodipodi:namedview
     id="namedview7"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:showpageshadow="2"
     inkscape:pageopacity="0.0"
     inkscape:pagecheckerboard="0"
     inkscape:deskcolor="#d1d1d1"
     inkscape:document-units="mm"
     showgrid="false"
     inkscape:zoom="0.60572619"
     inkscape:cx="178.29838"
     inkscape:cy="165.0911"
     inkscape:window-width="1920"
     inkscape:window-height="1027"
     inkscape:window-x="1920"
     inkscape:window-y="120"
     inkscape:window-maximized="1"
     inkscape:current-layer="layer1" />
  <defs
     id="defs2">
    <rect
       x="295.51306"
       y="774.27723"
       width="237.73118"
       height="234.42935"
       id="rect773" />
  </defs>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1">
    <g
       aria-label="ニコラ"
       id="text591"
       style="font-size:98.7997px;line-height:0;font-family:TakaoMincho;-inkscape-font-specification:'TakaoMincho, Normal';letter-spacing:0px;stroke-width:2.47">
      <path
         d="m 120.65121,36.783151 -4.43827,7.188064 q 4.39002,4.775962 6.6574,11.771057 2.26738,6.995096 2.26738,15.82339 0,8.152904 -1.97793,14.858548 -1.92968,6.705643 -5.78904,11.481605 -3.85937,4.775965 -9.60017,7.381035 -5.69256,2.60507 -13.170076,2.60507 -9.117745,0 -15.533937,-3.42519 l -3.762879,6.12674 -6.609159,-3.76288 4.341783,-6.995093 q -4.679477,-4.824204 -6.898611,-12.012268 -2.170892,-7.188064 -2.170892,-16.354052 0,-8.201146 1.833197,-14.858548 1.833198,-6.705643 5.644319,-11.433363 3.811121,-4.72772 9.600166,-7.284548 5.789045,-2.605071 13.652497,-2.605071 4.534752,0 8.394112,1.013083 3.90761,1.013083 7.09158,2.846281 l 3.85937,-6.126739 z M 73.470491,71.565662 q 0,6.319707 1.109567,11.385121 1.109567,5.017172 3.425185,8.780051 L 105.93738,46.142106 q -2.31561,-1.399019 -5.06541,-2.170891 -2.749797,-0.771873 -6.174982,-0.771873 -5.499592,0 -9.503681,1.977924 -4.00409,1.929681 -6.60916,5.596076 -2.60507,3.618153 -3.859363,8.876536 -1.254293,5.258382 -1.254293,11.915784 z m 42.163539,0 q 0,-12.398205 -4.2453,-19.682753 L 83.504835,97.230427 q 2.267376,1.350777 5.017172,2.074407 2.749797,0.723636 6.078497,0.723636 5.499596,0 9.455436,-1.929686 4.00409,-1.977924 6.56092,-5.644319 2.60507,-3.714637 3.81112,-8.97302 1.20605,-5.306624 1.20605,-11.915783 z"
         style="writing-mode:vertical-lr"
         id="path113" />
      <path
         d="m 108.83191,196.38827 -7.71873,-21.85365 H 74.821268 l -7.815211,21.85365 h -9.069503 l 25.905975,-70.81932 h 8.201147 l 25.954214,70.81932 z M 98.266899,166.57469 91.127077,146.12006 q -0.289452,-0.86836 -0.72363,-2.12265 -0.385936,-1.25429 -0.820115,-2.70155 -0.434178,-1.44726 -0.868357,-2.99101 -0.434178,-1.59199 -0.820114,-3.08749 -0.337694,1.4955 -0.771873,3.08749 -0.434178,1.59199 -0.868357,3.08749 -0.434178,1.44726 -0.868356,2.70155 -0.385937,1.20606 -0.675389,2.02617 l -6.995096,20.45463 z m -5.644318,-46.36061 h -5.789045 q -1.688471,-1.35077 -3.666395,-3.3287 -1.977924,-2.02616 -3.859363,-4.14881 -1.833198,-2.12265 -3.376943,-4.05233 -1.495503,-1.97793 -2.267376,-3.28046 v -1.01308 h 10.565007 q 0.771872,1.64023 1.784955,3.61815 1.061325,1.92968 2.170892,3.9076 1.109567,1.92969 2.267376,3.76288 1.157809,1.8332 2.170892,3.23222 z"
         style="writing-mode:vertical-lr"
         id="path115" />
      <path
         d="m 109.50729,314.26074 q 0,6.60916 -1.64022,11.77106 -1.59199,5.11366 -4.63124,8.63533 -2.99101,3.47342 -7.284549,5.30662 -4.293541,1.8332 -9.648408,1.8332 -4.96893,0 -9.165987,-1.8332 -4.197058,-1.8332 -7.284548,-5.30662 -3.039249,-3.52167 -4.775962,-8.63533 -1.688472,-5.1619 -1.688472,-11.77106 0,-6.56091 1.591988,-11.62633 1.640229,-5.11365 4.631236,-8.58708 2.991006,-3.52167 7.284548,-5.30662 4.341783,-1.8332 9.69665,-1.8332 4.96893,0 9.165987,1.8332 4.197058,1.78495 7.236307,5.30662 3.08749,3.47343 4.77596,8.58708 1.73671,5.06542 1.73671,11.62633 z m -37.049882,0 q 0,10.08259 3.280459,15.19625 3.3287,5.06541 10.757975,5.06541 7.429274,0 10.66149,-5.06541 3.280458,-5.11366 3.280458,-15.19625 0,-10.08258 -3.3287,-15.05151 -3.280458,-5.01717 -10.709732,-5.01717 -7.429275,0 -10.709733,5.01717 -3.232217,4.96893 -3.232217,15.05151 z m -0.04824,-40.71628 q 0,-2.60507 1.350777,-3.76288 1.350777,-1.15781 3.280459,-1.15781 1.881439,0 3.280459,1.20606 1.399019,1.1578 1.399019,3.71463 0,2.55683 -1.399019,3.76288 -1.39902,1.20605 -3.280459,1.20605 -1.929682,0 -3.280459,-1.20605 -1.350777,-1.20605 -1.350777,-3.76288 z m 18.090765,0 q 0,-2.60507 1.350777,-3.76288 1.350777,-1.15781 3.280459,-1.15781 0.916599,0 1.736713,0.28946 0.868357,0.28945 1.495503,0.9166 0.627147,0.5789 1.013083,1.4955 0.385937,0.9166 0.385937,2.21913 0,2.55683 -1.39902,3.76288 -1.399019,1.20605 -3.232216,1.20605 -1.929682,0 -3.280459,-1.20605 -1.350777,-1.20605 -1.350777,-3.76288 z"
         style="writing-mode:vertical-lr"
         id="path117" />
    </g>
  </g>
</svg>

t-oster, 2023-05-31:
You should use the Inkscape-Plugin, because the SVG-Parser in VisiCut has problems with Text-Elements. See the FAQ for more information https://github.com/t-oster/VisiCut/wiki/FAQ

nicolas-raoul, 2023-05-31:
The Inkscape-Extension (in the Extras menu) worked, thanks a lot!

It might be worth catching the exception to outputting this helpful tip when it happens. :-)

closed 2023-05-31 by nicolas-raoul

piccojer, 2025-07-08:
> You should use the Inkscape-Plugin, because the SVG-Parser in VisiCut has problems with Text-Elements. See the FAQ for more information https://github.com/t-oster/VisiCut/wiki/FAQ

Hi,

I have installed the inkscape plugin. But I am unsure how to "use" it. All it does, is give back the notification, that it is installed successfully. 

How do I actually use the inkscape plugin to import something into visicut?

mgmax, 2025-07-11:
The Inkscape extension is also explained in the user manual, which you can find in the Help menu or under this link: https://github.com/t-oster/VisiCut/wiki/VisiCut-manual