t-oster/VisiCut #681: [feature request] support for mapping based on group/layer name matching regex

aviraldg, 2023-08-12:
use case: Using Affinity Publisher's (but this should apply for other programs too) [Data Merge](https://affinity.help/publisher/en-US.lproj/index.html?page=pages/Advanced/dataMerge.html?title=Data%20merge) functionality to create templated copies of a design results in group names being different per-copy.

e.g. if the original group is called "group" before data merge, and there are 3 copies, then you might end up with "group1", "group2", and "group3"

TheAssassin, 2024-03-01:
Looks like a very specific problem of a very specific tool (that, honestly, I never had heard before). Might be easier to use some post-processing script (or pre, depending on your point of view) to alter the SVG to move those groups on separate layers, a feature that is already supported.

Adding such very niche features adds complexity and is not useful to the majority of users.

mgmax, 2024-03-03:
@aviraldg I would suggest that you use "Map by Color" so that you do not need to rely on the group names.

From what I have understood, your request is very complicated to implement (and then complicated to maintain). So I would decline it for now.

closed 2024-03-03 by mgmax