t-oster/VisiCut #682: Correct Arch Linux download

mgmax, 2023-08-20:
Now that the build and download server is working again (Thanks Thomas), one small update is necessary: Arch Linux package building was removed but old Arch packages are shown on the download site.

I would suggest that the "Arch Linux" tab on https://download.visicut.org/ is changed to show a message like:

Arch Linux users should build VisiCut themselves from the AUR https://aur.archlinux.org/packages/visicut-git according to the explanations in https://wiki.archlinux.org/title/Arch_User_Repository

TheAssassin, 2024-03-01:
I don't see Arch Linux on VisiCut at all. Do you mean it should be added?

Where is the source code of that page? Maybe it can be replaced with GitHub releases anyway with #670?

mgmax, 2024-03-02:
The original issue has been (more or less) resolved by removing the outdated Arch download tab.

The source code of that website is not published AFAIK. @t-oster maintains it.
Once #670 is done we could remove it.

closed 2024-03-02 by mgmax