t-oster/VisiCut #684: VisiCutMac-1.9-200-g5624a69b JAVA error

skywalkerdude, 2023-08-25:
I downloaded version 1.9-200 from https://download.visicut.org/, but I'm unable to open it on my mac. I get the error:
```
ERROR launching 'VisiCut'.

No suitable Java version found on your system!
This program requires Java 11 or later.
Make sure you install the required Java version.
```

I have Java 20:
```
java --version
openjdk 20.0.2 2023-07-18
OpenJDK Runtime Environment Zulu20.32+11-CA (build 20.0.2+9)
OpenJDK 64-Bit Server VM Zulu20.32+11-CA (build 20.0.2+9, mixed mode, sharing)
```

and my `JAVA_HOME` is set correctly. In fact, [Version 1.9.1](https://github.com/t-oster/VisiCut/releases/tag/1.9-71) works just fine. It looks like the newest 1.9-200 version is reading the java version incorrectly

mgmax, 2023-08-26:
This must be some Mac-specific issue for which I can't help too much because I don't have a Mac.

I assume you are running the Mac version and not the "Platform independent ZIP" version. We actually bundle a Java 11 in the Mac version and I don't understand why it is not detected by the universalJavaApplicationStub.
https://github.com/t-oster/VisiCut/blob/master/distribute/mac/VisiCut.app/Contents/MacOS/universalJavaApplicationStub
Maybe due to the switch to Apple Silicon?

Some things you could try:
Use the Platform independent ZIP version instead.
Maybe you want to try your luck with updating that stub file inside the application to a newer version available from the URL inside that file.
It can also help if you compare the folder contents of both versions to see what is different.

mgmax, 2023-08-29:
Can you try to move the folder and see what happens?

PR #686 (2023-09-12): Fix location of embedded JRE in macOS bundle
This was inadvertently broken in 4bd60b0. The JRE is intended to be installed under Contents/Plugins/JRE, but the refactoring inadvertently made it be installed under Contents/Plugins only. In turn, this meant the launched script could no longer find the JRE and gave an error instead of starting the application.

Fixes #684.
commit b8ba6089 (2023-09-12): Fix location of embedded JRE in macOS bundle

This was inadvertently broken in 4bd60b0. The JRE is intended to be
installed under Contents/Plugins/JRE, but the refactoring inadvertently
made it be installed under Contents/Plugins only. In turn, this meant
the launched script could no longer find the JRE and gave an error
instead of starting the application.

Fixes #684.
closed 2023-09-12 by t-oster