t-oster/VisiCut #689: Visicut COM port

ingacvinga, 2023-10-02:
I bought a Wainlux lasercutter model K6.  The software that came with the lasercutter (on USB stick) is not recognised by my computer at all, so I decided to give visicut a go. I am not a programmer and making this programme work is a hassle for me. I came so far that everything is set (at least I think so) but when I execute a cut it says error IOexception: no such port
[image]
 (i am attaching a picture).  can you help me what to write in COM port settings to make it work for windows 10? i found an article saying I should write COMx (where x needs to be changed to the real number of the connected COM Port). but how do i know this number? i just tried with COM1 and COM2 and it doesn't work. also my laser is connected to the computer via USB cable, not COM cable. when I plug it in the computer, I cannot find it in the devices and it says that the device I plugged in is malfunctioning. I really hope I can still make the lasercutter work. Am I missing some driver still? I really hope someone might help me with these questions, but I am an artist and I want to engrave my logo for my products and programming is really not my skill 🙁

Thank you for your help,

t-oster, 2023-10-02:
Hi,
if the COM port does not show up in windows device manager, then you're probably missing a driver for your device or your cable is broken.

ingacvinga, 2023-10-03:
Thank you for your answer. Would you know were could I find the correct driver?