t-oster/VisiCut #691: Visicut 1.8 fails to start on Ubuntu 22.04

jnweiger, 2023-11-09:
Should I try newer or older versions of Java?
```
pib@pib-VirtualBox:~$ visicut
Gtk-Message: 12:20:41.826: Failed to load module "canberra-gtk-module"
Exception in thread "AWT-EventQueue-0" java.lang.ExceptionInInitializerError
	at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:679)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:456)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:381)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:336)
	at com.t_oster.visicut.managers.PreferencesManager.getXStream(PreferencesManager.java:339)
	at com.t_oster.visicut.managers.PreferencesManager.loadPreferences(PreferencesManager.java:349)
	at com.t_oster.visicut.managers.PreferencesManager.getPreferences(PreferencesManager.java:287)
	at com.t_oster.visicut.gui.VisicutApp.initialize(VisicutApp.java:98)
	at org.jdesktop.application.Application$1.run(Application.java:170)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @228575c0
	at java.base/java.lang.reflect.AccessibleObject.throwInaccessibleObjectException(AccessibleObject.java:387)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:363)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:311)
	at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:180)
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:174)
	at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:59)
	... 22 more
```

jnweiger, 2023-11-09:
```
sudo apt-get install openjdk-{11,17,18}-{jdk,jre}
```
and playing with `update-alternatives --config java`

Result:
-> same crash with openjdk 17,18,19, 
-> visicut works fine with openjdk 11

mgmax, 2023-11-09:
Please try the java options listed here for a first diagnosis:
https://luppeng.wordpress.com/2023/04/01/use-openjdk-19-to-build-and-run-apache-atlas-v2-3-0/
```
--add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/java.lang.reflect=ALL-UNNAMED -Djdk.reflect.useDirectMethodHandle=false
```
There the error message is similar to ours.

jnweiger, 2023-11-10:
Workaround:
patch the /usr/bin/visicut wrapper to say 
```
# --add-opens ... and --Djdk.reflect... options are needed for openjdk-19 compatibility
java --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/java.lang.reflect=ALL-UNNAMED -Djdk.reflect.useDirectMethodHandle=false -Xms256m -Xmx1024m -jar "$DIR/Visicut.jar" --singleinstanceport $port "$@"

mgmax, 2023-11-11:
Good to see that we have a work-around.  I'm not sure why it broke, as we already apply a bunch of these "opens" settings in the pom.xml file. Visicut definitely worked with JDK17 at some point, I spent quite some time to make it work initially. As far as I remember, our Windows and Mac versions also ship a bundled JDK that is version 17 or similar. 

Please try to move all the "opens" stuff to the pom.xml file (just search for "opens" in the file). Everything there gets applied to all platforms and not just Linux. 

Then please try if the -D ... option is really needed. If yes then it probably needs to be added in three places - Linux, Mac and Windows Launcher.

mgmax, 2023-11-11:
Can you please double-check if it is really broken with Java 17?
Your commandline output in the issue report looks like from an older version. The latest VisiCut searches for the correct Java version and warns if it cannot find Java 11 or 17.
https://github.com/t-oster/VisiCut/commit/540595a1cdf42976efaa1c1b25dcbcdbb7530ff2

TheAssassin, 2023-12-25:
VisiCut works fine with the Java bundled in the AppImages, as far as I can tell, which is a prebuilt JVM found in https://github.com/adoptium/temurin17-binaries. I'd consider that the "reference environment" because the AppImage works on a majority of distributions.

The JAR built by `distribute.sh` also works (locally) on my workstation with OpenJDK 19. Same goes for https://download.visicut.org/files/master/All/VisiCut-1.9-209-g4df13952.zip. I checked the latter in an Ubuntu 22.04 VM as well. In both cases, I used the provided launch script.

mgmax, 2023-12-26:
I can't reproduce the issue:

- Start Ubuntu 22 from live ISO
- enable "universe" package source in "Applications and Updates" (needed for potrace)
- download VisiCut DEB
- open with "Software Installation" and install it
- Run visicut --> VisiCut works correctly and uses Java 11.
- `sudo apt install openjdk-19-jre`
- `VISICUT_JAVA=/usr/lib/jvm/java-19-openjdk-amd64/bin/java visicut` --> VisiCut works correctly and uses Java 19.
- same for Java 21

--> Closing as not reproducible.

closed 2023-12-26 by mgmax