t-oster/VisiCut #695: Wiki: Clarify adding kabeja dependency to local maven repo

pimsierhuis, 2023-12-05:
It seems that the dependency "kabeja" is not to be found in any operational maven repository. It would speed up people wanting to contribute to Visicut if some instructions how to put it in their local maven repo were given in the wiki. Here's the commands I used (after downloading from sourceforge, unpacking and cd'ing to lib-directory):

`
mvn install:install-file -Dfile=kabeja-0.4.jar -DgroupId=org.kabeja -DartifactId=kabeja -Dversion=0.4 -Dpackaging=jar
mvn install:install-file -Dfile=kabeja-svg-0.4.jar -DgroupId=org.kabeja -DartifactId=kabeja-svg -Dversion=0.4 -Dpackaging=jar
mvn install:install-file -Dfile=kabeja-xslt.jar -DgroupId=org.kabeja -DartifactId=kabeja-xslt -Dversion=0.4 -Dpackaging=jar
`

mgmax, 2023-12-05:
Thanks for your report. I am somewhat confused because since many many years ago we already use a workaround in our `pom.xml` file to ensure that kabeja is installed appropriately from a local file in the repository. So you should not have needed to do this manually. Either the old workaround broke or you did something different from what everyone else did when setting up the development environment :-)

Can you please write down what exactly you did (see https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started ) and where it fails? It would be best if you started fresh in a virtual machine (or another PC or some other reproducible environment).

Maybe we can at some point move from the Wiki page towards some kind of install script so that there is less copy-paste effort and so that the script can be automatically tested by the build server. 
See this example for Inkscape: https://gitlab.com/inkscape/inkscape-ci-docker/-/blob/master/install_dependencies.sh?ref_type=heads

pimsierhuis, 2023-12-05:
Sorry! My fault. I didn't need the detailed netbeans instructions (I used intellij) and therefore I didn't read through the mvn initialize step.

closed 2023-12-05 by pimsierhuis