t-oster/VisiCut #703: Silent install options

jger3110, 2024-02-28:
Hi there,

would it be possible to  release the next version for Windows with an added option to install the software silent? It seems the Windows installer is created with NSIS so an added option for the switch /S would be amazing. 
We are deploying the tool to a few PCs in our network using software deployment tools and not doing anything manual but instead using the unattended installation with the /S switch would by great.

Thanks in advance

Joerg

TheAssassin, 2024-03-01:
I have not had time to test this yet, but as far as I can see, NSIS should support this out of the box. What happens currently if you run the installer with that flag?

mgmax, 2024-03-02:
Somehow my reply by email got lost, reposting it again...

Hi,

that's a good idea. But good things will only happen if someone does them 😉

According to the internet, the /S switch should work out of the box. But due to whatever reason, the VisiCut installer just does nothing when called with /S.

Feel free to develop the needed changes. Here are some links to get started:

https://github.com/t-oster/VisiCut/blob/master/distribute/windows/installer.nsi

https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started
(Note that you can also use a Linux machine for building the Windows installer.)

NSIS silent mode:
https://nsis.sourceforge.io/Docs/Chapter4.html#silent

How to debug NSIS: https://nsis-dev.github.io/NSIS-Forums/html/t-325521.html

General NSIS programming advice: https://nsis.sourceforge.io/Docs/Chapter2.html

Maybe a first step would be to start from scratch with some NSIS MUI2 Example from the internet, see if /S works there, and if yes, then try to find out which difference to the VisiCut installer is relevant. If no, it's a bug in NSIS itself.

https://github.com/kichik/nsis/blob/master/Examples/Modern%20UI/Basic.nsi

Regards

Max

jger3110, 2024-03-07:
Sorry for the late reply, I answered by mail and just now checked to see that my reply is nowhere to be found...

By email I wrote:

NSIS states it accepts /S, /SD and /D as parameters for different aspects of the silent installation. Your setup does not accept any of those, it just starts the jnstaller GUI and awaits user input (Press Next to continue).
The expected behaviour would be for no GUI to show up and just install the software to the default path.
I had the chance of reading a little bit of the NSIS documentation and it says you would need to specifically add a section to your installer script.

This is the documentation I found.
https://nsis.sourceforge.io/Which_command_line_parameters_can_be_used_to_configure_installers%3F

Thanks for taking the time to look into this.

Greetings

Joerg

mgmax, 2024-03-07:
As I wrote above, you are invited to start investigating where the problem is, as probably you are currently the only one who needs it :-)

From a first look at the file, the VisiCut installer does nothing special, so I don't immediately see why the silent install does not work. The next steps would be to find out:
1. if the NSIS MUI2 example works correctly with /S
2. if yes, which part of the VisiCut NSIS file is problematic compared to that NSIS example? (try changing the working example step by step to make it more similar to VisiCut, until you find the point where it starts failing)
3. if no, how it is supposed to be done (ask on the NSIS forums for a working example, or try even simpler NSIS example files without MUI2)

mgmax, 2024-10-16:
Please try again with the latest version VisiCut-2.1-4 from https://github.com/t-oster/VisiCut/releases "Continuous build"

closed 2024-10-16 by mgmax

jger3110, 2024-10-18:
Confirmed, issue is recolved. /S installes the software silently. Thank you.