t-oster/VisiCut #704: Mapping in commandline?

valkyriesavage, 2024-02-29:
`$ ./VisiCut.MacOS --help

openjdk version "17.0.6" 2023-01-17
OpenJDK Runtime Environment Temurin-17.0.6+10 (build 17.0.6+10)
OpenJDK 64-Bit Server VM Temurin-17.0.6+10 (build 17.0.6+10, mixed mode)
Usage: visicut [-h|--help|-v|--version]
	 visicut [options] [<filename>]
	 visicut [options] --execute filename
Options are:
 --add (do not replace old file if VisiCut is already running)
 --material <materialname e.g. "Acrylic Glass 2mm">
 --laserdevice <laserdevice e.g. "Epilog ZING @ Miltons Office">
 --mapping <mapping e.g. "Cut">
 --total-height <Height in mm e.g. "2.5"> (only valid with --execute)
 --singleinstanceport <port> (Set port to check for running instances -- 0 to disable)
 --basepath <path> 	 Sets VisiCuts settings directory (default is $HOME/.visicut)
 --gtkfilechooser (experimental)`

However,

`./VisiCut.MacOS --mapping "Cut"
openjdk version "17.0.6" 2023-01-17
OpenJDK Runtime Environment Temurin-17.0.6+10 (build 17.0.6+10)
OpenJDK 64-Bit Server VM Temurin-17.0.6+10 (build 17.0.6+10, mixed mode)
Unknown command line option: --mapping
Use -h or --help for help`

Is it possible to call specific mappings from the commandline? And, if so, can it support something more complicated than "Cut"? (e.g., I am interested in mapping red strokes to Cut and blue strokes to Mark)

mgmax, 2024-03-03:
This is currently not implemented. I added an error message and some small improvements in Version 2.0-3. Due to issues with the build server it may take some days until you can download the version on download.visicut.org.

Workaround:
The code for --execute supports .PLF files. You can create these files with VisiCut - just load a file you want to cut, set up everything, and then use "File Save".  You can later modify this file from some outside script. Basically these are renamed ZIP files containing both the SVG file for lasercutting and a XML file with the material/mapping/... settings. So you can, e.g., build a Python script that unzips the .PLF, replaces the SVG file with the content you want and optionally makes other changes.

Long term solution:
Unfortunately, setting the mapping (especially, loading existing saved mappings) is the one part where the VisicutModel class is not cleanly separated from the GUI PropertyPanel. So some restructuring of the code is needed :-(

valkyriesavage, 2024-03-04:
Thanks! The workaround supports what I need for the moment, anyway.