t-oster/VisiCut #705: Replace `isCut` in profiles with color setting

TheAssassin, 2024-03-01:
Instead of configuring colors for cuts and engravings in the materials (non-cut vector profiles are considered engravings), I'd rather configure colors per configured profile. For instance, when configuring a cut, a rough engraving, a fine engraving and a mark (non-cut vector) profile, these could and should be assigned colors globally.

Of course, this could interfere with material colors, but honestly, I don't think that that conflict is much of an issue. One could remove this feature entirely or alternatively display the color in the sidebar to continue to allow the user to distinguish materials quickly. For instance, all wood colors will be some kind of brown-yellow-ish color anyway.

TheAssassin, 2024-03-01:
Another idea I just had which will be easier to configure but doesn't solve the core issue: add a color setting for non-cut vector profiles to the material settings.

mgmax, 2024-03-03:
In my opinion, VisiCut currently works correctly according to the "What you see is what you get" philosophy. What you are requesting looks more like "show the colors as in the original SVG" or "show wrong but very distinguishable colors", which are other philosophies. So, given a good use-case, we should add some switch to select how the colors are displayed. Just like in Google Maps where you can switch between "satellite view" or "artificial map view".

A good reason for the current philosophy is when you use the camera view: If you (vector-)engrave a black piece of acrylic, it doesn't make sense if the engraving is shown in black, because then you will have "black on black" which is hard to see. Any other engrave color than white will have similar problems, because you can buy acrylic also in other colors like dark-green.

Can you maybe give a concrete example for your use-case, where it is important to view the difference between different kinds of engravings in VisiCut?  It would be good if you can make a screenshot/photo of all three steps: the input SVG, of VisiCut's view, and of the finished workpiece.

TheAssassin, 2024-03-07:
I use a standard profile that I think most labs use: red for cutting, green for marking, blue for ignored elements, and the rest is engraved. Engravings are usually drawn by the user in black for the maximum contrast. Blue is hardly used.

It's hard for users to distinguish between those elements within VisiCut once a mapping was chosen because VisiCut basically re-colors some if not all elements. Even though non-cut vectors are now rendered thinner, it's not always easy for users to tell "oh, that's a mark". It's natural to think that those are cuts, especially when no actual cut is in the bed.

For this reason, I think "non-cut vector" should be treated as a separate kind of element just like vectors and engravings already are. I know that markings are technically vectors, and obviously they are already rendered differently. And while I personally got used to this quickly, users who do not use the lasercutter as often will have problems there.

I did not think about the camera part as of yet. But I don't think it really plays much of a role here. Or at least that's a different aspect. The colors you configure for a profile are usually picked to work with the background color (well, in my case I keep them the same for all materials and choose suitable (light) colors for the background). When using VisiCam, one can never predict what kind of image will be displayed. If you chose, e.g., black for raster profiles because the majority of material you sell is transparent acrylic and the user then enters a black sheet, you're in trouble, just as you described. It's unpredictable. Therefore, this should be discussed separately. As said, I could imagine removing the background color feature altogether.

I'll try to illustrate this with some pictures soon.