t-oster/VisiCut #709: Focus value not exported anymore after using execute

viralatadesbois, 2024-03-04:
Hi, 
We use a lot of thick material so we make the focus on the surface for engraving and have a focus value for each material set to minus half the thickness (for example for 8mm mdf the cut setting has a focus of -4mm)
We generaly export the gcode and send it to the laser cutter via its web interface (it's a Robotseed lasercutter with a smoothieboard). And it works fine. 
But to spare some times we also use the "execute" button from visicut to send the gcode directly. This is where we have a strange behavior:

- the first time we use "execute" everything is fine but then the focus value is not sent anymore, and even if we export the gcode after this first execute the focus value is not exported until we close and start again visicut.
We use visicut 1.9-200 on linux and windows 11 with the same behavior and it's the same on visicut 2.0-4 on linux.
Anybody with the same issue ? Any idea how to solve it ?

Thank you.

viralatadesbois, 2024-03-05:
[Visicut-offset-problem.zip](https://github.com/t-oster/VisiCut/files/14499466/Visicut-offset-problem.zip)

Thank you for your answer.
It happens fot what it seems everytime, wathever the offset (positive or negative) we use.
I joined our config, and 2 gcodes of the same file exported before and after the "execute" command and we see the line with G0 Z-3 disapeared.

mgmax, 2024-03-05:
oops... without even looking at your file, I now remember that I almost fixed that bug two years ago but never found time to finish that...

So the issue is in LibLaserCut, the library that provides the laser drivers of VisiCut. Moving the issue over to there.

See https://github.com/t-oster/LibLaserCut/pull/177 :
The currentFocus variable was not reset correctly, so sending the same job twice did sometimes not have the same content.

We should also add a test against such bugs in LibLaserCut. We already have some tests that check that the content is the same when the same job is sent twice . These were introduced in https://github.com/t-oster/LibLaserCut/commit/84435545e1bf4efcce0621ad0d91f964afa1e2db but specifically for the focus we don't do that yet.