t-oster/VisiCut #711: Add "frame"/"preview" feature

TheAssassin, 2024-03-07:
Partly related to #706.

VisiCut should be able to "draw" a frame on the bed around the current object(s). Basically a vector "cut" with a power of 0. I found such a feature while testing LightBurn lately and it's really, really useful for positioning.

For some controllers like Ruida, one could send the program and run it right away. For devices like the Epilog Zing, a special "preview" job could be sent which the user can then execute on the device.

Devices lacking a red dot feature could still benefit from this because the toolhead would move anyway. Depending on the size of the toolhead and its assembly, it still gives a rough estimate. Devices with an air assist nozzle close above the bed won't lose a lot of precision there.

TheAssassin, 2024-03-07:
Looking at https://github.com/t-oster/LibLaserCut/issues/158, I found the "move to position" feature which works similarly.

mgmax, 2024-03-08:
Same as "move to position", this feature could be handled without special "realtime hooks" by just sending an appropriate laser job. (0% power and slow speed? Or only MOVETO commands? As far as I know, MOVETO commands do not support a speed.)

mgmax, 2024-03-09:
As an interesting side node: The vector laserprofiles have a setting "use outline", however as far as I remember this applies to each object and not to the whole job. (I guess nobody uses that setting anymore...)

TheAssassin, 2024-03-09:
It might be worth removing obsolete features to clean up the code base a bit. But that's for another issue.

Sending a job would be the "fallback" solution. Some devices, e.g., the OMTech Polar, lack a user interface to select different jobs, so it's basically "operating by hope". It just has a single white button and an LED light, that's it. That's the motivation behind #706.

mgmax, 2024-03-10:
I would suggest starting with the "fallback" solution of sending a job. Then we have a starting point supported by all machines. For example, this could be a method in `LaserCutter` with a default implementation "send job with MOVETO rectangle" that can later be overridden for machines that have more appropriate commands.

mgmax, 2024-04-03:
Duplicate of https://github.com/t-oster/VisiCut/issues/181

closed 2024-04-03 by mgmax