t-oster/VisiCut #712: How to build both VisiCut and LibLaserCut at the same time in an IDE?

TheAssassin, 2024-03-07:
I try to use IntelliJ IDEA at the moment to work on VisiCut. While it works sufficiently well for VisiCut, I struggle to build LibLaserCut along with the main project. Maven links VisiCut against the installed JAR in the Maven cache.

I don't quite see why LibLaserCut is a separate project anyway, but that's for another discussion. I first need to solve this major inconvenience in order to work on the Ruida driver.

TheAssassin, 2024-03-07:
I just stumbled upon https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started. It looks like Netbeans provides some IDE-specific feature to enable such a workflow. But I think the build system itself should support such functionality. E.g., in CMake, there is `add_subdirectory`.

TheAssassin, 2024-03-07:
I found a way to add the project to IDEA following [this guide](https://www.jetbrains.com/help/idea/delegate-build-and-run-actions-to-maven.html#link_unlink_maven_project). I'll add some instructions to the wiki once I figured it all out.

TheAssassin, 2024-03-07:
I added some basic instructions to https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started. I also cleaned up some links.

closed 2024-03-07 by TheAssassin

mgmax, 2024-03-08:
If it makes sense, you could also add an IntelliJ project file to the repository.

Does running and debugging work correctly from IntelliJ? At least Netbeans could not pick up the `AddOpens` from the `pom.xml`, so loading settings caused weird Exceptions until I added this manually to NetBeans' configuration: https://github.com/t-oster/VisiCut/commit/6866709e0e0221f1a814729272f4ae853e12334f