t-oster/VisiCut #713: Change from boolean "origin bottom left" to enum to support all four corners

TheAssassin, 2024-03-08:
Some Ruida controlled devices have their bottom on the top right corner instead of the more traditional left side. VisiCut already supports the top and bottom left, but lacks support for the other corners. For instance, the device I try to make VisiCut work on just executes everything mirrored right now when "origin bottom left" is not set.

I intend to replace the very limiting boolean with an enum to support all four corners for now.

t-oster, 2024-03-08:
Hi,
I remember working on similar problems on the branches https://github.com/t-oster/VisiCut/tree/feature-originbottom and https://github.com/t-oster/VisiCut/compare/master...feature-633-import-bottom-left. That's probably not exactly what you're trying to get, but may be worth looking at.
However the first branch never got finished and the second never got merged, so this is just to inform you.

mgmax, 2024-03-15:
Please note that there are several different aspects:

1. Displayed coordinates in Visicut GUI: Where the VisiCut UI displays the (0,0) point in the rulers and in the Position panel. This was tried in https://github.com/t-oster/VisiCut/compare/master...feature-originbottom .
2. Default placement of files when the SVG page size is not equal to the lasercutter bed size. This was tried in https://github.com/t-oster/VisiCut/compare/master...feature-633-import-bottom-left .
3. Default "Origin point" when the user does not set a relative position via Right click - set start point (*)
4. Transformation inside the LibLaserCut device driver from LibLaserCut coordinates (always defined: as 0,0 at absolute top left,  X towards right, Y towards bottom) to device command coordinates. This also includes what the driver does if the users sets a start point by Right-click. For Ruida devices this was discussed in https://github.com/t-oster/LibLaserCut/pull/206 but is currently waiting for answers and more thoughts

(*) this is only relevant for machines where the user can force that an absolute job is still executed as a relative job (or vice versa). For example, Epilog Zing uses stepper motors and has a button to temporarily switch off the stepper motors so that the user can move the axis by hand.