t-oster/VisiCut #716: Consider removing vectorization feature(s)

TheAssassin, 2024-03-11:
Continuation from https://github.com/t-oster/VisiCut/pull/710#issuecomment-1989364070.

I'm not sure how much these features are used and whether they provide reasonable results, but personally, I think we're better off using dedicated tools for the job. For instance, Inkscape provides a sophisticated vectorization feature.

mgmax, 2024-03-15:
I'd suggest we wait a year and see if someone picks it up and improves it. Else, I think it is okay to drop so that we have less maintenance burden.

Currently, there are some random errors (timing issues?) when dragging the sliders, and the usability could also be improved.

mgmax, 2025-04-08:
the year has passed without any improvements, so the feature can now be removed. Help is welcome.