t-oster/VisiCut #718: Reproducible build, checksum verification of external dependencies

mgmax, 2024-03-24:
Some parts of the build process download external dependencies without verifying a checksum. This means that if any of our external dependencies or their download server is compromised, the next VisiCut build will automatically pick this up. 

Good example:
Download of JDK for Windows/Mac setup. https://github.com/t-oster/VisiCut/blob/8797012c2feb29f909edd2f448dced487ed70a5a/distribute/distribute.sh#L111

Insufficiently checked parts:
- Maven dependencies. This is tricky, see
  - https://stackoverflow.com/questions/3307146/verification-of-dependency-authenticity-in-maven-pom-based-automated-build-syste
  - https://issues.apache.org/jira/browse/MNG-6026
  - Possible solution: https://github.com/chains-project/maven-lockfile
- pyuploadtool:  https://github.com/t-oster/VisiCut/blob/8797012c2feb29f909edd2f448dced487ed70a5a/.github/workflows/github-build.yml#L63 (introduced in #710)
- appimagecraft pip git (only weak SHA1 hash, and only a short part of it) https://github.com/t-oster/VisiCut/blob/8797012c2feb29f909edd2f448dced487ed70a5a/distribute/Dockerfile#L16
- downloads by appimagecraft itself, e.g., https://github.com/TheAssassin/appimagecraft/blob/6b36fda05d50b356024bd3b12ca426ba73fe334b/appimagecraft.yml#L13 (only relevant for AppImage build, not for other package types)

TheAssassin, 2024-03-24:
I think you might be a little too concerned in this regard. In the software industry "downloaded from a static GitHub URL" is even sufficient. It's rather unlikely that someone would manage to work around all the security features (TLS, HSTS even). I can relate to the wish for a static build so that new ones don't break things (even if that is rather uncommon in the world of AppImages, for instance), but checking hashes is a bit over the top.

mgmax, 2024-03-24:
Regarding "static GitHub URL": Anyone with sufficient access to the dependency repository could change the content of the release artifact, and then we would silently download and run that new content. No need to hack GitHub :-)

TheAssassin, 2024-03-24:
Sure, but then again we trust these projects to have proper access control in place.