t-oster/VisiCut #720: Visicut shows extra padding around object

jnweiger, 2024-04-03:
In inkscape I have drawn a numeral 3 (using hershey 1 line font)
The numeral is ca 4cm high, the viewbox fits very tightly.
For the screenshot, I pressed CTRL-A (select all) to demonstrate the effective bounding box: 

[image]

When this SVG is sent to visicut (either via extension or save and load) it is shown with a huge padding.
The start point of the laser cannot be set anywhere near the object.

[image]

Expected behaviour:
* bunding box of the object in visicut is the exact bounding box. 
* viscut should not get confused by nested groups that stack several scalings and translations.

[image]
[bb33.svg.zip](https://github.com/t-oster/VisiCut/files/14849293/bb33.svg.zip)

mgmax, 2024-04-26:
Thanks for the good bugreport! Can you try to further simplify the example file?

Currently the file has five levels of groups, transforms etc., which will be quite difficult to debug.

jnweiger, 2024-04-26:
I've simplified the SVG file to one path with a scale transform. No nested groups needed.
The scale is a downscale for a factor of 1000, to emphasize the effect.

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg width="40mm" height="30mm" viewBox="0 0 40 30">
  <path
     d="M 0,0 H 20000 L 5000,20000 h 10000"
     style="fill:none;stroke:#0000ff;stroke-width:100"
     transform="scale(0.001,0.001)"
     id="path4" />
</svg>
```

This figure Z is constucted with dimensions of 20000. The scale reduces this to 2 cm x 2 cm. 
That much works identical in inkscape and in visicut.

The stroke width is set to 100; in inkscape the scale is correctly applied, resulting in an effective stroke width of 0.1mm
In visicut, the stroke  remain unaffected by the scale. That is a bug.
The bounding box computation includes sufficient padding to keep the entirety of a 100mm wide stroke inside the bounding box. We get 50 mm added to each side.
[image]

[image]

jnweiger, 2024-04-26:
BTW: In inkscape there is a switch, visual bouning box / geometric bouning box, I ususall choose the latter, so that stroke widths do not distort object sizes.
Visicut seems to have visual bounding box hardcoded, otherwise this bug would have no effect.
This also means, that thickening the lines in inkscape, moves the drawing in visicut by half the added stroke width. 
That is probably a surprise to most users...

mgmax, 2024-04-27:
With #727 it should hopefully be better. I didn't test it in detail, maybe it is still wrong for some special case.

Now the stroke width is used correctly, although we need it only for engraving ("visual bounding box") and not for cutting ("geometric bounding box"). Please create a separate issue for that.

commit cc3689d4 (2025-04-08): Fix #720: Bounding box calculation for paths with transform

For `<path transform="..." style="stroke-width:...">`, VisiCut ignored the `transform` when calculating the effective stroke width that is used to determine the bounding box.

As a result, some files showed way too large bounding boxes and could not be moved fully to the top left.
closed 2025-04-08 by mgmax