t-oster/VisiCut #731: Visicut doesn't align layers from svg correctly

Roycinger, 2024-09-11:
Hi,
I was preparing a drawing in Inkscape like this:
[image]
After importing in Visicut, the 2 layers (outline and text) are distorted / misplaced:
[image]

Is there anything I'm missing?
This should be 2 jobs for engraving and cutting

t-oster, 2024-09-13:
could you upload the SVG, so we can investigate?

Roycinger, 2024-10-05:
> could you upload the SVG, so we can investigate?

Sorry for getting back late, I missed the email notification.
I think I trashed the initial file from the screenshot but I still got another similar file with quite the same issue.
I'll append it to this post
[image]
Thanks for having a look!

t-oster, 2024-10-05:
@Roycinger this file works as expected on my machine (R)
[image]
[image]

Roycinger, 2024-10-05:
Looks like this for me:
[image]
In your first screenshot, there's a small portion of the text as well right on the top

t-oster, 2024-10-06:
You are right. The XML Structure is:
[image]

and visicut seems to interpret the text-nodes correctly and the tspan nodes in the text as additional text on the wrong position. I will investigate further.

t-oster, 2024-10-06:
until then it will help if you convert the text to objects in inkscape (which should be done by the visicut inkscape plugin imho)

t-oster, 2024-10-06:
If we just ignore tspan element, this particular file works. However since tspan elements can have their own attributes (https://developer.mozilla.org/en-US/docs/Web/SVG/Element/tspan) we should handle them correctly.
We probably should treat the text-node the same as a group node (aka ignore the node itself, only use the attributes in the hierarchy) but we have to wrap all text in text nodes without tspans in some artifical tspan object.

Maybe we should collect a few test files for this, before start implementing.