t-oster/VisiCut #732: VisiCut compatibility with FluidNC

TimHall82, 2024-09-11:
Hello everyone, just wondering if VisiCut has compatibility with FluidNC firmware?

If not, is this something that can easily be added?

Thanks,

Tim

t-oster, 2024-09-13:
Hi, I don't know the FluidNC firmware, but the website says it can use GRBL compatible G-Code, so you may be lucky with any of the g-code drivers.
If you need something special, there is a tutorial here https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started and more specifically here https://github.com/t-oster/VisiCut/wiki/Developing-a-new-Lasercutter-Driver

mgmax, 2024-10-16:
FluidNC should be usable with the Grbl driver once #734 is fixed.

mgmax, 2024-11-10:
See the latest discussions in #734