t-oster/VisiCut #734: G-Code Export without space

paranoid64, 2024-10-14:
Hello,

the g-code that is exported is without spacing.  

I haven't found anything on how to fix this with settings?

```
G21
G90
M3
G0X43.942000Y4.318000S0
G1X48.9204Y4.318S0F6000
G1X49.0728Y4.318S600
G1X49.1236Y4.318S0
G1X49.3268Y4.318S600
G1X49.3776Y4.318S0
G1X49.5808Y4.318S600
G1X49.6316Y4.318S0
G1X49.8348Y4.318S600
G1X49.8856Y4.318S0
G1X49.9364Y4.318S600
G1X54.9148Y4.318S0
G0X42.011600Y4.368800S0
G1X46.99Y4.3688S0F6000
G1X47.1424Y4.3688S600
G1X47.1932Y4.3688S0
G1X47.3964Y4.3688S600
G1X47.4472Y4.3688S0
G1X47.6504Y4.3688S600
G1X47.7012Y4.3688S0
G1X47.9044Y4.3688S600
G1X47.9552Y4.3688S0
G1X48.1584Y4.3688S600
```

I always use the G-Code export because FluidNC opens its own access point and everything works via webUI without a cable connection. Unfortunately, the G-code only works with this when you first search and insert space between the command and the coordinates. Otherwise the program would be perfect for me.

t-oster, 2024-10-14:
Hi,
could you post a screenshot of your Laser-Cutter settings in VisiCut. It is probably not much work to add a setting for adding a space to the selected driver. Also I guess the upload could be implemented directly in a new driver, which inherits the currently used one.

paranoid64, 2024-10-14:
Hi,

[image]

So there is a space in there.

mgmax, 2024-10-16:
The Grbl GCode driver removes spaces because they are not needed for Grbl. I doubt that the space is really the problem. Even the test files in the FluidNC repository have some commands without spaces:
https://github.com/bdring/FluidNC/blob/main/FluidNC/src/tests/parsetest.nc#L6

I don't understand what you mean about LAOS-HPC. The LAOS driver does not use GCode but some other format:
https://github.com/t-oster/LibLaserCut/blob/master/test-output/de.thomas_oster.liblasercut.drivers.LaosCutter.out
Maybe you are looking for the "Generic GCode" driver?

Best guess for the actual solution:

Space is not the problem. The problem is the line ending. When saving the file with your text editor, the line ending is changed from CR to LF or CRLF and then it works.

For Grbl we use CR (I don't know why). Grbl treats CR and LF the same:
https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/protocol.c#L110
FluidNC only supports LF and CRLF:
https://github.com/bdring/FluidNC/blob/f9016b7468f881143c1c3716b426edde9de6f19e/FluidNC/src/InputFile.cpp#L22-L28

So as a workaround we could change the Grbl driver to use LF, and then hope that's enough so that it is compatible with FluidNC.

mgmax, 2024-11-10:
A draft version with CRLF instead of LF is available for download here under "artifacts".
https://github.com/t-oster/VisiCut/actions/runs/11764084933?pr=736

Please give feedback if that works.

paranoid64, 2024-11-13:
Hi, works for me. But I don't think FluidNC was the problem, because I only exported the G-code and did the same in NCviewer (Online G-Code Simulator). Nothing was visible there either. Maybe it's the computer and I have to test it again on the same computer.

mgmax, 2024-11-13:
It may be the same issue, that the online simulator does not like CR line endings (which is quite exotic, because Windows uses CRLF and Linux LF. Only ancient Mac software used CR.)

closed 2024-11-24 by mgmax

TimHall82, 2025-01-08:
Still seem to have compatibility issues with FluidNC:

When executing jobs, I get an error stating that the firmware did not respond with "ok" and instead I get co-ordiantes (G0X105.460800Y69.39800S0). I can open the xml file for the lasercutter in the users/specialised/lasercutters directory and change the following text:

<waitForOKafterEachLine>true</waitForOKafterEachLine>
 
Changing this from "true" to "false" makes the machine move to the job co-ordinates and begin processing the job, however the machine stops after processing around 2 seconds of code. The entirety of generated Gcode therfore is not processed by the firmware.

mgmax, 2025-01-10:
Which option did you change from "true" to "false"? It looks like GitHub removed some part of your comment due to special characters like < or >. Try to use the "Code" markup button in the comment field.

It seems like FluidNC has echo activated, which means that it repeats everything it receives back to the sender. This should normally not be the case and it should be gone if you restart FluidNC.
http://wiki.fluidnc.com/en/features/serial_terminals#advanced-terminal-mode