t-oster/VisiCut #735: Impossible to load file on Ubuntu 24.04

TadaNoButa, 2024-11-09:
This is weird: I've used Visicut 2.1 a month ago and everything was ok. Now, presumably after some updates of Ubuntu 24.04, when I click the open file buttton, the file chooser dialog opens and I can navigate thru the folders, but it doesn't show any file (.svg nor .plf). Then, when I click the cancel button, the dialog chooser closes, but the app is totally unresponsive and I have to kill it.

I've then tried with the AppImage, instead of .deb, tried the continuous build AppImage, the old 1.9 .deb: it's all the same. That is sure some external library problem, but I can't figure out that it is. I've tried to monitor what the console says:  not much actually, except it's using openjdk version "11.0.24" 2024-07-16.

I guess I'll use another computer with an older Ubuntu for a while to be able to use Visicut.

mgmax, 2024-11-09:
For me everything, works in a fresh Ubuntu 24.04 VM with all updates and VisiCut 2.1 installed as .deb.

Is there anything different about your installation?
Can you please try to reproduce the problem on another PC (or in a VM)?

TadaNoButa, 2024-11-11:
I was about to check this on a fresh 24.04 install and I checked once again on the computer I'm normally using and: it works again !

I'm really sorry and I feel dumb. But I really had this problem the other day and made many reboots with always the same bad result. So, I don't know what to say else than was working, then had this problem, then is working again. I don't think I made a Ubuntu update within those two days, but I couldn't tell for sure.

I'm very happy it works again and so sorry to have bothered you for seemingly nothing. And thank you the developers for this great application !

closed 2024-11-11 by mgmax