t-oster/VisiCut #737: Execution problem on mac

Alireza-Safdari-Khosroshahi, 2024-11-18:
After starting the VisiCut i got an error. 
[image]
After start the VisiCut from the content folder and prepareing the execution got this one. 
[image]

t-oster, 2024-11-18:
Looks as if the purejavacomm library is not ARM compatbile, at least the version we are using.

t-oster, 2024-11-18:
Are you using a laser cutter over USB or over network?

Alireza-Safdari-Khosroshahi, 2024-11-19:
I’m using a K3 laser cutter connected via USB.

t-oster, 2024-11-20:
[test.zip](https://github.com/user-attachments/files/17831528/test.zip)
can you try to run this version? I replaced the purejavacomm library with 
```
<dependency>
    <groupId>org.bidib.jbidib</groupId>
    <artifactId>jbidibc-purejavacomm</artifactId>
    <version>2.0.28</version>
</dependency>
```
(but I did not check where it is from, nor what is inside, so be cautious)

Alireza-Safdari-Khosroshahi, 2024-11-20:
First of all, thank you for your support.
I have checked the version, but unfortunately, it didn’t work. It seems there is an issue with opening the TTY. 

[image]

t-oster, 2024-11-20:
Well at least this looks different. Now this may be a permission issue, but I am not an osx user