t-oster/VisiCut #741: XML Deserialization Vuln

royblume, 2025-01-27:
Identified a vulnerability in the loadPlfFile method of VisicutModel.java within the VisiCut application version 2.1. The application utilizes the java.beans.XMLDecoder class to deserialize XML data extracted from user-provided PLF files without proper validation or sanitization. This vulnerability allows an attacker to craft a malicious .plf file containing specially crafted XML content that, when processed, leads to arbitrary code execution on the underlying system, which could result in denial of service, privilege escalation, and information disclosure.

t-oster, 2025-01-28:
Hi, thanks for reporting. Do you have a quick fix?

mgmax, 2025-01-29:
Luckily  like we only use it in one place (except for an unused import):
https://github.com/t-oster/VisiCut/blob/87daf9a5f665debb8d27bce116022602b7c9e6a8/src/main/java/de/thomas_oster/visicut/VisicutModel.java#L525-L526
Everywhere else we use XStream. 

And even luckier, we only use it to decode `java.awt.geom.AffineTransform`, which is only a fancy way to store six double values:
```
<java version="1.6.0_31" class="java.beans.XMLDecoder">
<object class="java.awt.geom.AffineTransform">
<array class="double" length="6">
<void index="0">
<double>0.3527777777777778</double>
</void>
<void index="3">
<double>0.3527777777777778</double>
</void>
<void index="4">
<double>-207.8862438374972</double>
</void>
<void index="5">
<double>-73.74667636085336</double>
</void>
</array>
</object>
</java>
```
Note that zeros are omitted from the XML.

So it should be possible to decode by dumb string matching, or to craft a regex that limits the XML to precisely the structure we expect. String matching is probably easier because we can just ignore the header and we can't make mistakes in the regexp.

t-oster, 2025-01-29:
Thanks for checking. This sounds like an easy task, maybe I can fix it tomorrow.

commit e862ecda (2025-01-30): Replace XMLDecoder with a specialized regex-based solution in order to prevent security vulnerabilities on untrusted input. Fixes #741
closed 2025-01-30 by t-oster