t-oster/VisiCut #743: Add "Germany - Berlin - xHain Hack+Makespace" URL to downloadn recommended settings

neurove, 2025-02-09:
Please add the settings for the xHain Hack+Makespace (Berlin, Germany) to the recommended settings.
The file can be found at  https://github.com/xHain-hackspace/visicut-settings/blob/main/xhain-laser-settings.vcsettings
Thanks!

TheAssassin, 2025-02-09:
You can add it yourself to https://github.com/t-oster/VisiCut/blob/master/src/main/java/de/thomas_oster/visicut/misc/LabSettings.java and make a pull request to get it in. 

I would however recommend you to restructure your repository, i.e., you probably just want to commit the settings files in your `~/.visicut` settings directory directly. I suppose nobody has given it a try yet and you may be able to use the exported `.vcsettings` file you pushed to your repository (that's just a zip archive of your `~/.visicut` anyway, if you use the raw download link GitHub provides, it should work), but most people just use the method I described instead. It's quite pointless to commit an archive to Git, it'll just inflate the repository's size, and you cannot directly inspect the files from a Git repository browser such as GitHub.

You can check out the example repositories linked in `LabSettings.java`. As Git (like Forgejo, Gitea, GitLab, ...) support exporting a branch's contents to a zip archive, they can be imported in VisiCut like any other `.vcsettings` file. If you want to provide users a way to import the files manually, too, you can give them that same link.

neurove, 2025-02-10:
Thanks for the suggestion.
We implemented a similar workflow.
I also added the setting based on your suggestion and created a pull request: https://github.com/t-oster/VisiCut/pull/744

closed 2025-02-10 by neurove