t-oster/VisiCut #746: List of laser machines disappears

andrewerrington, 2025-02-19:
Hello,

Just started using VisiCut. Very nice! One issue I noticed was that if you have more than one laser then a drop-down list appears at top right of the main window. If you _had_ more than one laser and delete all but one then the drop down list disappears. This is alarming to new users. I suggest that the list is always present, even if it contains only one item.

Thank you.

mgmax, 2025-04-08:
Thanks for your request. I understand what you want. However, the current behavior is intentional, to make the UI as simple as possible for people that have only one lasercutter. As a workaround you can add a second lasercutter named e.g. "Do not use - Old laser". If you have a better suggestion that is easy to implement and still keeps the UI simple for people with just one laser, please let me know.

closed 2025-04-08 by mgmax

andrewerrington, 2025-04-09:
Ok. Thanks for reading my report and responding, however I disagree with your conclusion. I also disagree with the decision to make this behaviour intentional. What you have done is make the UI change its appearance based on the number of laser cutters defined, without stating anywhere that this happens. I also can't believe you are suggesting a workaround, when the problem itself can be, and should be, addressed. In my opinion you should be aiming for consistency, and adhere to the principle of least surprise.

Adding a second "Do not use" entry as a workaround is a poor suggestion, since it changes the UI (from no list, to a list appearing), and, more importantly, if you have two lasers, for example, one named "My laser", which you use, and "Do not use - old laser", which you don't, you have now made a useless UI widget (i.e. the only reason for the second entry is to make the list appear, but it's now possible to mistakenly select the "wrong" option, which is only there to make the list appear). Furthermore, if you delete "My laser", the list disappears, and the software is now using the remaining entry ("Do not use - old laser"), which it probably shouldn't be using, and there is no way to see that this is the case.

I already posted "a better suggestion that is easy to implement and still keeps the UI simple for people with just one laser": I suggest that the list is always present, even if it contains only one item.

Specifically, by keeping the list always present, even if it contains only one item:
UI never changes- things don't magically and inexplicably disappear
UI shows which laser is in use. If there is only one, then it's reassuring to see it.
The existence of the list also helps to make the UI more discoverable, in that it implies that it may be possible to define more than one laser.
It's impossible to leave behind a single laser marked "Do not use - old laser" and not know about it.
It removes a tiny piece of code which decides whether or not to display the list (and other code which must handle these dynamic UI changes), thus leading to less code. Less code means less to maintain, and fewer bugs.

Thank you for coming to my TED talk.