t-oster/VisiCut #747: Java Deserialization Vulnerability

Gelcon, 2025-02-26:
There is a Java deserialization vulnerability in VisiCut 2.1. You can get the PoC resources at https://www.cve.org/CVERecord?id=CVE-2024-51365.

closed 2025-02-26 by t-oster

Gelcon, 2025-02-26:
Dear mgmax,

  When the vulnerability is triggered, there occurs a stack overflow error, which will cause visicut to fail to function.
This is the fault of CVE. So I will correct the description of this CVE.

Best Regards,
Gel

mgmax, 2025-03-24:
@Gelcon can you please correct the description of the CVE? Thanks.

Gelcon, 2025-03-25:
> [@Gelcon](https://github.com/Gelcon) can you please correct the description of the CVE? Thanks.

Yeah, I have submit my request of updating the description of this CVE ID.
[image]

commit a08a84fa (2025-02-26): Bump XStream Version. Fixes #747