t-oster/VisiCut #750: List of Laser Machines disappears (reprise)

andrewerrington, 2025-04-10:
I don't seem to be able to re-open this issue, which was closed yesterday, so I am re-creating it.

> Ok. Thanks for reading my report and responding, however I disagree with your conclusion. I also disagree with the decision to make this behaviour intentional. What you have done is make the UI change its appearance based on the number of laser cutters defined, without stating anywhere that this happens. I also can't believe you are suggesting a workaround, when the problem itself can be, and should be, addressed. In my opinion you should be aiming for consistency, and adhere to the principle of least surprise.
> 
> Adding a second "Do not use" entry as a workaround is a poor suggestion, since it changes the UI (from no list, to a list appearing), and, more importantly, if you have two lasers, for example, one named "My laser", which you use, and "Do not use - old laser", which you don't, you have now made a useless UI widget (i.e. the only reason for the second entry is to make the list appear, but it's now possible to mistakenly select the "wrong" option, which is only there to make the list appear). Furthermore, if you delete "My laser", the list disappears, and the software is now using the remaining entry ("Do not use - old laser"), which it probably shouldn't be using, and there is no way to see that this is the case.
> 
> I already posted "a better suggestion that is easy to implement and still keeps the UI simple for people with just one laser": I suggest that the list is always present, even if it contains only one item.
> 
> Specifically, by keeping the list always present, even if it contains only one item:
> UI never changes- things don't magically and inexplicably disappear
> UI shows which laser is in use. If there is only one, then it's reassuring to see it.
> The existence of the list also helps to make the UI more discoverable, in that it implies that it may be possible to define more than one laser.
> It's impossible to leave behind a single laser marked "Do not use - old laser" and not know about it.
> It removes a tiny piece of code which decides whether or not to display the list (and other code which must handle these dynamic UI changes), thus leading to less code. Less code means less to maintain, and fewer bugs.
> 
> Thank you for coming to my TED talk. 

 _Originally posted by @andrewerrington in [#746](https://github.com/t-oster/VisiCut/issues/746#issuecomment-2788109922)_

andrewerrington, 2025-04-10:
I've just installed VisiCut on someone else's machine, and I am reminded why this issue is so bad.

Initially, the drop-down list appears at the top right of the UI, saying something like "Add a laser", I can't recall exactly. So, users will add a laser, and then _the drop-down list disappears!_ This is truly awful.

This is in MainView.java, around line 595, and easy to fix.

    //hide lasercutter combo box if only one lasercutter available
    if (this.laserCutterComboBox.getItemCount() == 1)
    {
      this.laserCutterComboBox.setSelectedIndex(0);
      this.laserCutterComboBox.setVisible(false);
      this.jLabel9.setVisible(false);
    }
    else
    {
      this.laserCutterComboBox.setVisible(true);
      this.jLabel9.setVisible(true);
    }

The same problem is present around line 530:

    //hide material combo box if only one material available
    boolean materialUiVisible = true;
    if (this.materialComboBox.getItemCount() == 1 && MaterialManager.getInstance().getAll().get(0).getMaterialThicknesses().size() == 1)
    {
      this.materialComboBox.setSelectedIndex(0);
      materialUiVisible = false;
    }

Why? You have only defined one material, but we're not going to tell you what it is. Changing the UI based on arbitrary conditions is very bad, and leads to confusion. Please fix it.

t-oster, 2025-04-10:
When visicut was created, we had FabLabs in mind, where admins would configure visicut and users would start working with it without prior contact. We wanted to show the user the least amount of UI possible.

I agree with you that for a home user this may be confusing, but I still think the FabLab case is a valid point.

We could make it visible per default and add a setting to hide the choices when there is only one option, however this would add yet another setting...

andrewerrington, 2025-04-11:
I disagree entirely, which is why I opened an issue. If the UI has a combo box initially _it should not disappear!_ It should only disappear when an action is taken to _make it disappear._ Not only that, but I think that using the combo box as a message box to say "Please install a driver" is itself poor UI practice, and now you need special code to tell the difference between a combo box containing a single configured laser, and a combo box containing a single entry that is not a laser, it's a message.

I strongly believe the combo box should have zero entries on a fresh installation, and always be visible. Now you don't need special code to check if the single entry in the list is a laser, or a message. If the list is empty (zero entries) no laser has been defined. If the list has one item, it's a single laser. If it has more than one then there are two lasers. You could also keep track by using the list index (which you have to remember anyway, to remember the last used laser). It's -1 if there's nothing in the list (I think) and 0..n-1 for n lasers.

If you want to hide things from the user (because you don't want people changing things in a fablab, for example, or you want to hide things so they are not distracting or initimidating) then there should be a menu option like "Hide options", or "Expert mode", or something. If necessary this could pop up a dialog box with a passcode to allow it to be changed. Or it could be a value in a config file somewhere that can only be changed by editing the file.

I am currently helping a new user get up to speed with laser cutting. They can't afford the price of the, er, popular commercial offering, so I thought VisiCut would be ideal. Having watched their consternation when things changed without them doing some action to make it change I decided to submit a bug report. Yes, I could gloss over it, tell them not to worry, and, since they won't change it again, pass it off as a non-issue, but I feel strongly that this is not the right way to do things. Hence raising the issue. Twice. It should perhaps be in "Discussions", but I think it is definitely an issue.

Actually, I probably should have read the closed issues first. It seems that I am not the first to comment on this. It was raised five years ago:
https://github.com/t-oster/VisiCut/issues/360

t-oster, 2025-04-11:
#360 pretty much covers the issue. There are just two distinct opinions for different use-cases.

If you submit a pull request which
- changes the default behavior to always show the comboboxes (lasercutter+material)
- adds a settting to hide comboboxes with just one element

that would be ok with me.

andrewerrington, 2025-04-12:
I'm sure you would. I was sort of hoping that one of your 56 contributors would see this issue and agree, and fix it.

I've just had to figure out why it crashed on startup today, but didn't yesterday. I couldn't see the cause exactly, but I deleted the hidden files in the top-level of the home directory and this fixed it.

I have another issue which I'll open shortly.